bkerler/oppo_decrypt #2: Can u make video tutorial how to decode ofp file , ,  for newbie

0x1999, 2018-10-22:


francko76, 2018-10-30:
I tried to use ops_decrypt and It say this:

> Detected AES: a27940
> Found MAlloc: a36fd2
> 
> 
> 
> [+] Seek: 0xde9ce200
> [+] New DLength: 0x92880
> [+] New DLength: 0x92880
> 
> [+] ReadFileLength: 0x126b3
> [+] Subsize: 0x126b3
> [+] New ReadFileLength: 0x92880
> 
> [+] ReadFileLength: 0x92880
> 
> [+] AES-Length: 0x92880
> [+] AES-EDX: d41860b3
> Data dump Data :
>            0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F  0123456789ABCDEF
> 00000000  7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00  .ELF............
> Decrypted data was written to 'decrypted.data'.
> Writing data.
> 

It produces a file called decrypted.data about 587KB.

How to fully extract the ops ?

bkerler, 2020-04-18:
Use commit bfb6499 and ofp_extract.py

closed 2020-04-18 by bkerler