bkerler/oppo_decrypt #5: Unbrick oneplus 5t with save userdata

panty90, 2019-08-09:
I need to unbrick my oneplus 5t without loss my user data. All oneplus 5t unbrick tools have firmware in one file .ops. I need extract it and then I will try this guide to save my data. https://forum.xda-developers.com/oneplus-3t/how-to/unbrick-tool-3t-unbrick-data-babloo-t3737791
Please, help me!

btuhin17, 2019-08-09:
Whats you actually problem please explain

btuhin17, 2019-08-09:
But its not possible to safe your user data..sorry..all unbrick tools
method your data will be erase..after flash

btuhin17, 2019-08-09:
Your bootloader is unlocked or lock

panty90, 2019-08-09:
No. I find two guides for oneplus one and for oneplus 3t where userdata was saved.

panty90, 2019-08-09:
Bootloader is lock

panty90, 2019-08-09:
One of it is where https://forum.xda-developers.com/oneplus-3t/how-to/unbrick-tool-3t-unbrick-data-babloo-t3737791

btuhin17, 2019-08-09:
Are you indian??

btuhin17, 2019-08-09:
Or speak hindi or any indian languages

panty90, 2019-08-09:
In this guide for oneplus 3t firmware have separated files: boot, recovery and etc. In all unbrick tools for oneplus 5t firmware is in one file .ops. I need to extract it and after that change process loading without  erase userdata.
No)

btuhin17, 2019-08-09:
Yes...only OnePlus 3 & 3t firmware is separate part by part..but after oreo
firmware.. OnePlus combined there firmware in .ops file..its not possible
to extract (any normal user)only can OnePlus official developer or any
hardcore software programmer can decrypt this firmware.. becouse this
prosser is very complicated..

panty90, 2019-08-09:
So I described the problem here, where they offer codes for unpacking .ops files. Oneplus support says that not possible too but I know another. It their falt, that my phone is bricked after last update. Files very important for me and I can't erase it.

btuhin17, 2019-08-09:
Then it not possible to recover without erase your data...

btuhin17, 2019-08-09:
Or search on YouTube or Google "how to backup my phone data from edl 9008
mode

closed 2020-02-07 by bkerler