bkerler/oppo_decrypt #7: Help Please ! a fix needed for newbies like me

Ldev007, 2020-02-24:
Hello ! your scripts are just awesome ! and helped me alot already but I'm stuck at an stage which is "OFP Extraction". The thing is I don't have any idea about <startoffset> or <length> I tried researching but couldn't understand at all. Pardon my dumbness.

So, as an solution I thought to ask you out like what about if you can create another script which could just extract the whole .ofp file or you can also give options within the script itself to choose between complete extraction or offset specific extraction ? whatever you think about it please reply it's ok if it's not possible but please do reply.

Thanks,
Ldev

nathanvdh, 2020-03-05:
This a very specific set of very useful tools - both this and the ozip decrypter, but the average user has not really any documentation to go on. So you end up with all these "HELP ME PLEASE!!" issues. A short wiki or more detailed readme would be greatly appreciated.

bkerler, 2020-03-07:
The ofp decrypter no longer works, that's why it's not maintained and documented right now.

closed 2020-03-07 by bkerler