bkerler/oppo_decrypt #15: Is there any possibility to decrypt Oppo MTK ofp files?

shadabmozaffar, 2020-07-08:
Hi ViperBJK!

Thank you for your nice piece of code to extract ofp file. I have successfully converted it to delphi. 

As I can see, it only supports QC firmware. I could not find any way to find page size sign in all MTK firmware I have. Also random decryption did not help.

Can you please shed some light or help me to work on MTK firmware decryption?

Br,

bkerler, 2020-08-26:
Yes. It will be supported shortly. Consider donating as soon as it's up :)

closed 2020-08-26 by bkerler

shadabmozaffar, 2020-08-26:
Hi!

I will be more than happy to do that. Please share a payment method.

Br,
Shadab

Sandeepa167, 2021-03-20:
> Hi ViperBJK!
>
> Thank you for your nice piece of code to extract ofp file. I have successfully converted it to delphi.
>
> As I can see, it only supports QC firmware. I could not find any way to find [...]

hi ,

can you share your delphi converted code ?

thanks !