bkerler/oppo_decrypt #18: errors while extracting .ofp file of realme 3

vishalk17, 2020-08-09:
#A  
`  File "ofp_extract.py", line 153
    print(f"Extracting {wfilename}")
                                  ^
SyntaxError: invalid syntax`

#B  after commenting line no . 153 

`Unknown pagesize. Aborting`

can you plz do support this device too../

firmware link :  https://androidfilehost.com/?fid=8889791610682909823

bkerler, 2020-08-09:
Yeah I will have a look next week ;)

vishalk17, 2020-08-10:
> […](#)

I forget to tell you that.. realme 3 is an mediatek device.

Thanking You.

paindustry, 2020-08-24:
> […](#)

> > […](#)
> 
> I forget to tell you that.. realme 3 is an mediatek device.
> 
> Thanking You.

I try to run the code for RMX1941 and it doesnt work. Probably because It was also using DownloadTool.exe(MTK) not MSMDownloadTool.exe(QC)
since the tool was made by same company i wonder if it used the same generatekey for both tools

vishalk17, 2020-08-24:
> I try to run the code for RMX1941 and it doesnt work. Probably because It was also using DownloadTool.exe(MTK) not MSMDownloadTool.exe(QC)
> since the tool was made by same company i wonder if it used [...]

probably different key or master key for different .ofp files using by realme. although @bkerler  said that he'll take a look.. so there is still hope for mtk. ;)

bkerler, 2020-08-25:
Yeah, the algo is completely different. Already decrypted a small portion of the ofp file. Working on it.

vishalk17, 2020-08-25:
> Yeah, the algo is completely different. Already decrypted a small portion of the ofp file. Working on it.

Noice..

bkerler, 2020-08-26:
I've fully reverse engineered it ... it was a real pain to do so. Would be happy if someone would give me a donation for it :)
Code will be uploaded in a few minutes.

vishalk17, 2020-08-27:
yupp, sure...
........................

i m getting this error whenever executing script for mtk

i m using : 
                  ubuntu : 20.04 LTS server
                  python3 version : 3.8.2-0ubuntu2
                  python crypto : pycrypto-2.6.1

`vishalk17@temp:~/ofp/oppo_decrypt$ python3 ofp_mtk_decrypt.py ../RMX*/*.ofp extract/
Detected prjinfo:184433
Detected cpu:6771
Detected flash:emmc
Writing "DAA_AUTH_FILE" as "extract//auth_sv5.auth"...
Writing "EFUSE_CONFIG" as "extract//18601-efuse_oppo6771.xml"...
Traceback (most recent call last):
  File "ofp_mtk_decrypt.py", line 155, in <module>
    main(filename,outdir)
  File "ofp_mtk_decrypt.py", line 134, in main
    data=aes_cfb(aeskey,aesiv,encdata,True)
  File "ofp_mtk_decrypt.py", line 39, in aes_cfb
    plaintext = cipher.decrypt(data)
  File "/home/vishalk17/.local/lib/python3.8/site-packages/Crypto/Cipher/blockalgo.py", line 295, in decrypt
    return self._cipher.decrypt(ciphertext)
ValueError: Input strings must be a multiple of the segment size 16 in length
`

vishalk17, 2020-08-27:
Thanks ..solved .. issue was with pycrypto package .. i have uninstalled it then install pycryptodome==3.4.3 ( pip3 install pycryptodome==3.4.3) and files has been extracted successfully. I'll considered your paypal acc. for donation.

bkerler, 2020-08-27:
Thanks. With commit 59ea7fe it should work with pycrypto as well.

vishalk17, 2020-08-27:
> Thanks. With commit [59ea7fe](https://github.com/bkerler/oppo_decrypt/commit/59ea7fe4a1ada44acca2c881650725c513f774d3) it should work with pycrypto as well.

your paypal donation link is dead. Please update it so that i can request other users too for donation.

closed 2020-08-27 by vishalk17

paindustry, 2020-08-27:
Thanks it worked with "RMX1941EX_11_A.10_190507_c80b7906" looks like everything has been extracted but i can't found the DA_FILE (DownloadAgent.bin)