bkerler/oppo_decrypt #19: TypeError: 'iv' is an invalid keyword argument for this function

kirtangajjar, 2020-08-12:
```
Traceback (most recent call last):
  File "ofp_extract.py", line 252, in <module>
    main()
  File "ofp_extract.py", line 182, in main
    pagesize,key,iv,data=generatekey2(filename)
  File "ofp_extract.py", line 92, in generatekey2
    pagesize,data=extract_xml(filename,key,iv)
  File "ofp_extract.py", line 121, in extract_xml
    dec=aes_cfb(data,key,iv)
  File "ofp_extract.py", line 134, in aes_cfb
    ctx = AES.new(key, AES.MODE_CFB, iv=iv, segment_size=128)
  File "/home/kirtan/.local/lib/python3.6/site-packages/Crypto/Cipher/AES.py", line 95, in new
    return AESCipher(key, *args, **kwargs)
  File "/home/kirtan/.local/lib/python3.6/site-packages/Crypto/Cipher/AES.py", line 59, in __init__
    blockalgo.BlockAlgo.__init__(self, _AES, key, *args, **kwargs)
  File "/home/kirtan/.local/lib/python3.6/site-packages/Crypto/Cipher/blockalgo.py", line 141, in __init__
    self._cipher = factory.new(key, *args, **kwargs)
TypeError: 'iv' is an invalid keyword argument for this function
```

kirtangajjar, 2020-08-12:
Solved. The issue was package conflict with `pycrypto`. I uninstalled it, and installed `pycryptodom`

closed 2020-08-12 by kirtangajjar

All3xJ, 2024-04-23:
> Solved. The issue was package conflict with `pycrypto`. I uninstalled it, and installed `pycryptodom`

pycryptodome*