bkerler/oppo_decrypt #22: run command but no output

siddhrsh, 2020-08-27:
PS E:\New folder> python3 ofp_mtk_extract.py RMX1941EX_11_A.28.ofp
PS E:\New folder>

bkerler, 2020-08-27:
Hi, as stated in the readme, you need to also give the name of the directory to extract to:
in wsl: python3 ofp_mtk_decrypt.py RMX1941.ofp extractdir

Also if you use spaces in the names, you need to use quotationmarks as well:

python3 ofp_mtk_extract.py RMX1941EX_11_A.28.ofp "E:\New folder"

siddhrsh, 2020-08-27:
> Hi, as stated in the readme, you need to also give the name of the directory to extract to:
> in wsl: python3 ofp_mtk_decrypt.py RMX1941.ofp extractdir
> 
> Also if you use spaces in the names, you need to use quotationmarks as well:
> 
> python3 ofp_mtk_extract.py RMX1941EX_11_A.28.ofp "E:\New folder"

PS E:\New folder> python3 ofp_mtk_extract.py RMX1941EX_11_A.28.ofp "E:\New folder"
PS E:\New folder>
also i tried ofp_mtk_decrypt.py

siddhrsh, 2020-08-27:
tried via ubuntu 
python3 ofp_mtk_decrypt.py RMX1941EX_11_A.28.ofp extract
Unknown key. Please ask the author for support :)

siddhrsh, 2020-08-27:
> Hi, as stated in the readme, you need to also give the name of the directory to extract to:
> in wsl: python3 ofp_mtk_decrypt.py RMX1941.ofp extractdir
> 
> Also if you use spaces in the names, you need to use quotationmarks as well:
> 
> python3 ofp_mtk_extract.py RMX1941EX_11_A.28.ofp "E:\New folder"

tried via ubuntu
python3 ofp_mtk_decrypt.py RMX1941EX_11_A.28.ofp extract
Unknown key. Please ask the author for support :)

paindustry, 2020-08-27:
it worked with RMX1941EX_11_A.10
maybe for versions vA.20 and above have used a different key.

> > Hi, as stated in the readme, you need to also give the name of the directory to extract to:
> > in wsl: python3 ofp_mtk_decrypt.py RMX1941.ofp extractdir
> > Also if you use spaces in the names, you need to use quotationmarks as well:
> > python3 ofp_mtk_extract.py RMX1941EX_11_A.28.ofp "E:\New folder"
> 
> error on windows, wsl2 & ubuntu
> python3 ofp_mtk_decrypt.py RMX1941EX_11_A.28.ofp extract
> Unknown key. Please ask the author for support :)

siddhrsh, 2020-08-27:
> it worked with RMX1941EX_11_A.10
> maybe for versions vA.20 and above have used a different key.

tried on a17 too same issue
too many changes from a10 to a28 so i want the latest i have extracted recovery.cpio.gz too but there is no sbin/recovery

AungKhantMin, 2020-10-23:
if i am not wrong you can only extracted the key from mtk download tool

AungKhantMin, 2020-10-23:
> > it worked with RMX1941EX_11_A.10
> > maybe for versions vA.20 and above have used a different key.
> > > > Hi, as stated in the readme, you need to also give the name of the directory to extract to:
> > > > in wsl: python3 ofp_mtk_decrypt.py RMX1941.ofp extractdir
> > > > Also if you use spaces in the names, you need to use quotationmarks as well:
> > > > python3 ofp_mtk_extract.py RMX1941EX_11_A.28.ofp "E:\New folder"
> > > 
> > > 
> > > error on windows, wsl2 & ubuntu
> > > python3 ofp_mtk_decrypt.py RMX1941EX_11_A.28.ofp extract
> > > Unknown key. Please ask the author for support :)
> 
> tried on a17 too same issue
> too many changes from a10 to a28 so i want the latest
> 
> i have extracted recovery.cpio.gz too but there is no sbin/recovery waiting for @bkerler reply

if i am not wrong you can only extracted the key from mtk download tool

bkerler, 2020-10-23:
I will have a look ....

AungKhantMin, 2020-10-24:
> I will have a look ....

I created a pull request for this issue @bkerler

bkerler, 2020-10-24:
Fixed with commit 5696b05ac3b07921ab2f698f9aae600c9805f3f0 thanks to AungKhantMin

FreyjadourF, 2020-10-26:
I'm having the same problem, tried extracting ofp from CPH2127PU_11_A.23_200826_6aade87b.ofp

closed 2020-10-26 by bkerler