bkerler/oppo_decrypt #27: Oppo A5: the decrypted imgs still encrypted

kwaiching, 2021-01-14:
decrypted file:
https://androidfilehost.com/?fid=11410963190603873017

original ofp:
链接: https://pan.baidu.com/s/1wbEidozxFNfryax4vKwcSA  密码: o9ob

the ofp can be decrypted, but the imgs still encrypted:
$ file boot.img 
boot.img: data

$ file opporeserve2.img 
opporeserve2.img: data

$ file recovery.img 
recovery.img: PGP Secret Key -

the normal boot:
$ file boot_11.1.053_st.img
boot_11.1.053_st.img: Android bootimg, kernel (0x8000), ramdisk (0x1000000), page size: 4096, cmdline (console=ttyMSM0,115200,n8 androidboot.console=ttyMSM0 earlycon=msm_serial_dm,0xc170000 androidb)

kwaiching, 2021-01-28:
Normal boot.img:
ANDROID!Xï  € €Ø           ð€  €      J                 console=ttyMSM0,115200,n8 androidboot.console=ttyMSM0 androidboot.hardware=qcom user_debug=30 msm_rtb.filter=0x237 ehci-hcd.park=3 androidboot.bootdevice=7824900.sdhci lpm_levels.sleep_disabled=1 earlycon=msm_hsl_uart,0x78B0000 vmalloc=300M androidboot.usbconfigfs=true loop.max_part=7 buildvariant=user 

The boot.img from ofp file:
UFIÿÓâº=¸À¼b9XpÈ„MåÞ„àá ðE›Wx7h!Ò†í*¹Á[)ÕìçZÑÞ*RQ¯Þ7j?øÜ:k¡console=ttyMSM0,115200,n8 androidboot.console=ttyMSM0 androidboot.hardware=qcom msm_rtb.filter=0x237 ehci-hcd.park=3 lpm_levels.sleep_disabled=1 androidboot.bootdevice=7824900.sdhci earlycon=msm_serial_dm,0x78af000 ramoops.mem_address=0x91400000 ramoops.mem_size=0x100000 ramoops.record_size=0x20000 ramoops.dump_oops=0 buildvariant=user               

It seems that the header was destroyed

hwangjr, 2021-03-18:
do you have any idea to solve this error? I want to extra the boot.img and patch with magisk, but it shows:
```
Unsupported/Unknown image format
```
so, the boot img maybe broken

kwaiching, 2021-03-18:
> do you have any idea to solve this error? I want to extra the boot.img and patch with magisk, but it shows:
> 
> ```
> Unsupported/Unknown image format
> ```
> 
> so, the boot img maybe broken

No until the boot.img was correctly fixed, and the MsmDownloadTool was Cracked.
I tried QPST, however it didn't work

hwangjr, 2021-03-18:
okay, thanks. I already install magisk trough other way...

anujchirde, 2021-04-11:
How did you install magisk the other way , please tell us about your other way.
Since my phone realme 2 is :
SAR : system as root 
Ramdisk : no (so no patching boot.img)
Twrp : not supported 

So only logical way for me is to patch recovery.img but can't since it is still encrypted

hwangjr, 2021-04-11:
I flash twrp and install magisk.

bkerler, 2021-05-05:
I will have a look.

bkerler, 2021-05-05:
It looks like they have (again) added new security. to be reversed ...

bkerler, 2021-05-06:
Fixed with commit 9a52620

kwaiching, 2021-05-17:
> Fixed with commit [9a52620](https://github.com/bkerler/oppo_decrypt/commit/9a526205d9db00d1a6f2301ea3c027c1fefd2bd4)

thanks a lot

closed 2021-05-19 by bkerler