bkerler/oppo_decrypt #28: Error on hashes. File might be broken ! Crypt

rebainoor, 2021-01-26:
Extracting prog_emmc_firehose_8953_ddr.mbn
Extracting dynamic_nvbk.bin
Error on hashes. File might be broken ! Crypt
Extracting static_nvbk.bin
Error on hashes. File might be broken ! Crypt
Extracting persist.img
Error on hashes. File might be broken ! Crypt
Extracting emmc_fw.bin
Error on hashes. File might be broken ! Crypt
Extracting marketname.xml
Error on hashes. File might be broken ! Crypt
Extracting opporeserve2.img
Error on hashes. File might be broken ! Crypt
Extracting sbl1.mbn
Error on hashes. File might be broken ! Crypt
Extracting rpm.mbn
Error on hashes. File might be broken ! Crypt
Extracting tz.mbn
Error on hashes. File might be broken ! Crypt
Extracting devcfg.mbn
Error on hashes. File might be broken ! Crypt
Extracting sbl1.mbn
Error on hashes. File might be broken ! Crypt
Extracting rpm.mbn
Error on hashes. File might be broken ! Crypt
Extracting tz.mbn
Error on hashes. File might be broken ! Crypt
Extracting devcfg.mbn
Error on hashes. File might be broken ! Crypt
Extracting adspso.bin
Error on hashes. File might be broken ! Crypt
Extracting sec_smt.dat
Error on hashes. File might be broken ! Crypt
Extracting emmc_appsboot.mbn
Error on hashes. File might be broken ! Crypt
Extracting emmc_appsboot.mbn
Error on hashes. File might be broken ! Crypt
Extracting NON-HLOS.bin
Error on hashes. File might be broken ! Crypt
Extracting dtbo.img
Error on hashes. File might be broken ! Crypt
Extracting vbmeta.img
Error on hashes. File might be broken ! Crypt
Extracting boot.img
Error on hashes. File might be broken ! Crypt
Extracting recovery.img
Error on hashes. File might be broken ! Crypt
Extracting logo.bin
Error on hashes. File might be broken ! Crypt
Extracting DRIVER.ISO
Error on hashes. File might be broken ! Crypt
Extracting cache.img
Error on hashes. File might be broken ! Crypt
Extracting mdtp.img
Error on hashes. File might be broken ! Crypt
Extracting lksecapp.mbn
Error on hashes. File might be broken ! Crypt
Extracting cmnlib_30.mbn
Error on hashes. File might be broken ! Crypt
Extracting cmnlib64_30.mbn
Error on hashes. File might be broken ! Crypt
Extracting keymaster64.mbn
Error on hashes. File might be broken ! Crypt
Extracting lksecapp.mbn
Error on hashes. File might be broken ! Crypt
Extracting cmnlib_30.mbn
Error on hashes. File might be broken ! Crypt
Extracting cmnlib64_30.mbn
Error on hashes. File might be broken ! Crypt
Extracting keymaster64.mbn
Error on hashes. File might be broken ! Crypt
Extracting dpAP.mbn
Error on hashes. File might be broken ! Crypt
Extracting system.img
Error on hashes. File might be broken ! Crypt
Extracting vendor.img

prithivirajasingh, 2021-03-09:
I have the same issue for
system.img
vendor.img
opporeserve2.img

bkerler, 2021-03-20:
Duplicate of issue #29

closed 2021-03-20 by bkerler