bkerler/oppo_decrypt #31: Super image didn't get extracted on Realme 8 Pro OFP

TheMalachite, 2021-04-03:
OFP zip: https://www.mediafire.com/file/041upspf79y26v3/RMX3081export_11_A.24_202103160933_GsmMafia.Com.rar/file

Output: https://del.dog/ifywexafun.txt

Other partitions got extracted except super.img

rokibhasansagar, 2021-04-05:
### Link: https://ghostmirror.hitesh920.workers.dev/RMX3081export_11_A.27_202103290141.ofp

### When using `pycryptodome` without `pycrypto`

```bash
Traceback (most recent call last):
  File "/home/phoenix_dumper/./utils/oppo_decrypt/ofp_qc_extract.py", line 9, in <module>
    from Crypto.Cipher import AES
ImportError: cannot import name 'AES' from 'Crypto.Cipher' (unknown location)
```

### When using `pycrypto` along with `pycryptodome`

```bash
Traceback (most recent call last):
  File "/home/phoenix_dumper/./utils/oppo_decrypt/ofp_qc_extract.py", line 362, in <module>
    main()
  File "/home/phoenix_dumper/./utils/oppo_decrypt/ofp_qc_extract.py", line 238, in main
    pagesize,key,iv,data=generatekey2(filename)
  File "/home/phoenix_dumper/./utils/oppo_decrypt/ofp_qc_extract.py", line 93, in generatekey2
    pagesize,data=extract_xml(filename,key,iv)
  File "/home/phoenix_dumper/./utils/oppo_decrypt/ofp_qc_extract.py", line 123, in extract_xml
    dec=aes_cfb(data,key,iv)
  File "/home/phoenix_dumper/./utils/oppo_decrypt/ofp_qc_extract.py", line 136, in aes_cfb
    ctx = AES.new(key, AES.MODE_CFB, iv=iv, segment_size=128)
  File "/usr/local/lib/python3.9/site-packages/Crypto/Cipher/AES.py", line 95, in new
    return AESCipher(key, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/Crypto/Cipher/AES.py", line 59, in __init__
    blockalgo.BlockAlgo.__init__(self, _AES, key, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/Crypto/Cipher/blockalgo.py", line 141, in __init__
    self._cipher = factory.new(key, *args, **kwargs)
TypeError: 'iv' is an invalid keyword argument for this function

```

#### In both time, `Crypto` package was installed

zacharee, 2021-04-28:
It looks like the script just doesn't even try to handle super extraction.

MlgmXyysd, 2021-05-16:
Same OnePlus 9 Pro ColorOS OFP

MlgmXyysd, 2021-05-16:
https://github.com/bkerler/oppo_decrypt/pull/34

closed 2021-05-16 by bkerler