bkerler/oppo_decrypt #35: Crash when repack OnePlus ops file

MlgmXyysd, 2021-05-24:
```batch
python3 opscrypto.py encrypt extract --projid=19805 --firmwarename=19805 --savename=19805.ops

Oneplus CryptTools V1.3 (c) B. Kerler 2019-2021
MIT License
----------------------------
Traceback (most recent call last):
  File "opscrypto.py", line 697, in <module>
    main()
  File "opscrypto.py", line 630, in main
    item, pos = copyitem(item, directory, pos, wf)
  File "opscrypto.py", line 517, in copyitem
    pos += rlen
TypeError: unsupported operand type(s) for +=: 'int' and 'NoneType'
```
I tried to add these code after line 517:
```python
if rlen == None:
            rlen = 0
```
Then ops can generate, but cannot load.
[image]

MlgmXyysd, 2021-05-24:
@bkerler

bijleeeeeeeee, 2021-05-24:
@bkerler - I have the same error while trying to encrypt OnePlus ops file

bkerler, 2021-06-03:
I will have a look :)

bkerler, 2021-06-03:
Solved with commit 821e0ee

MlgmXyysd, 2021-06-03:
works well :)

closed 2021-06-03 by MlgmXyysd