bkerler/oppo_decrypt #38: Unknown key. Please ask the author for support :)

ShinoNuma, 2021-06-01:
Hi @bkerler,

I've recently got the firmware file (.ofp) for the Oppo watch (41mm) and since I've successfully unlock the bootloader I wanted to extract the boot.img from the firmware in order to pach it with magisk.

But when I run ofp_qc_extract.py and ofp_mtk_decrypt.py I have the following error

ofp_mtk_decrypt.py
```
b'ab3f76d7989207f2' b'2bf515b3a9737835'
Unknown key. Please ask the author for support :)
```

Then for ofp_qc_extract.py 

```
Unknown key. Aborting
```

Do you have any idea?

bkerler, 2021-06-03:
Can you upload the msmdownloadtool and ofp for it ?

ShinoNuma, 2021-06-03:
Hey!

Yes, sure.

Here is the link:

```
https://drive.google.com/file/d/1AxDmw2OLeNlvGUa5oO4FSudgEYNpN0sA/view?usp=sharing
```

Let me know if you need anything else.

bkerler, 2021-06-03:
Solved with commit 821e0ee

ShinoNuma, 2021-06-03:
Awesome!

Thanks for all!

closed 2021-06-03 by ShinoNuma

MlgmXyysd, 2021-06-04:
> Hey!
> 
> Yes, sure.
> 
> Here is the link:
> 
> ```
> https://drive.google.com/file/d/1AxDmw2OLeNlvGUa5oO4FSudgEYNpN0sA/view?usp=sharing
> ```
> 
> Let me know if you need anything else.

Hi, can you get OPPO Watch China firmware? may I can do something crossflash

ShinoNuma, 2021-06-04:
Hmmm not sure about it, I've created this gist file :
```
https://gist.github.com/ShinoNuma/c4746c3a93928433200f16f8ecf2da49
```

Hope it will help you.