bkerler/oppo_decrypt #41: New Split Super img decryption issues

sarthakroy2002, 2021-06-13:
I have contacted you through a tweet for that split super img decryption. So here are the ofp of new realme UI 2 (colors 11) ofps.... which have split super imgs...

Qcom: https://fileload.coloros.com/489556RMX3081export_11_A.28_202104090212.zip [REALME 8 PRO]

MTK: https://fileload.coloros.com/489478RMX3191export_11_A.10_202103300234.zip [REALME C25]

MlgmXyysd, 2021-06-13:
simg2img cannot merge?

sarthakroy2002, 2021-06-13:
> simg2img cannot merge?

These arent any sparse img tho...some encrypted split imgs

techyminati, 2021-06-22:
> I have contacted you through a tweet for that split super img decryption. So here are the ofp of new realme UI 2 (colors 11) ofps.... which have split super imgs...
>
> Qcom: [...]

+1 , Same issues with me , super images are splitted in various small/small chunks. can't merge them via simg2img.

Ofp url ( realme C25 - given above by sarthak ) , realmeUi 2.0 /ColorOS 11

qiurigao, 2021-07-03:
[super_map.csv.txt](https://github.com/bkerler/oppo_decrypt/files/6758018/super_map.csv.txt)
super images are splitted cmd？
[image]

qiurigao, 2021-07-03:
nv_id,nv_text,super_0_path,super_1_path,super_2_path
10010111,ALLNET,super.0.d7972dc6.img,super.1.109ca5f2.img,super.2.895ec818.img
00000000,ALLNET,super.0.d7972dc6.img,super.1.109ca5f2.img,super.2.895ec818.img

MlgmXyysd, 2021-07-03:
> nv_id,nv_text,super_0_path,super_1_path,super_2_path
> 10010111,ALLNET,super.0.d7972dc6.img,super.1.109ca5f2.img,super.2.895ec818.img
> 00000000,ALLNET,super.0.d7972dc6.img,super.1.109ca5f2.img,super.2.895ec818.img

These are in ProFile.xml too

qiurigao, 2021-07-03:
How to use it so that super.img can be used @MlgmXyysd

qiurigao, 2021-07-03:
[image]

MlgmXyysd, 2021-07-03:
> How to use it so that super.img can be used @MlgmXyysd

> Qcom: https://fileload.coloros.com/489556RMX3081export_11_A.28_202104090212.zip [REALME 8 PRO]

just `wsl simg2img super.0.0e6049ae.img super.1.3712d40c.img super.2.9ae7d6a8.img super.img`
[image]

MlgmXyysd, 2021-07-03:
> > simg2img cannot merge?
> 
> These arent any sparse img tho...some encrypted split imgs

nope, I just downloaded your file and merged them

techyminati, 2021-07-03:
> > > simg2img cannot merge?
> > 
> > 
> > These arent any sparse img tho...some encrypted split imgs
> 
> nope, I just downloaded your file and merged them

Cool.

qiurigao, 2021-07-04:
@MlgmXyysd 每个转换出来了三个9.5gb的super.img 这三个文件怎么合并呢，提取其中的分区都是不完整的，不知道为啥，要联合使用，单独用来不了机

MlgmXyysd, 2021-07-04:
> @MlgmXyysd 每个转换出来了三个9.5gb的super.img 这三个文件怎么合并呢，提取其中的分区都是不完整的，不知道为啥，要联合使用，单独用来不了机

https://github.com/bkerler/oppo_decrypt/issues/41#issuecomment-873375685

qiurigao, 2021-07-04:
simg2img还能这么用，哪个版本

MlgmXyysd, 2021-07-04:
> simg2img还能这么用，哪个版本

I don't know which version it is

techyminati, 2021-07-24:
well yeah i got a fix now for this issue, just do this:

` sudo apt install simg2img `
` simg2img <spilt super1> <split super 2> ..... ( do like this for all spillted super images ) super.img ( type super.img in very last ) `

this works

eg; ` simg2img super.0.cc7f050e.img super.1.146b7f3b.img super.2.424660d8.img super.2.43c2aa98.img supe
r.2.5a9584e4.img super.2.6e6033c5.img super.2.82604042.img super.2.9a8d28fe.img super.2.a8472824.img super.img `

Thanks !

huckleberrypie, 2021-11-18:
> well yeah i got a fix now for this issue, just do this:
>
> `sudo apt install simg2img` `simg2img <spilt super1> <split super 2> ..... ( do like this for all spillted super images ) super.img ( type [...]

Do I have to combine all super IMGs in the case of MTK firmware such as this?
https://fileload.coloros.com/503830RMX2020export_11_C.07_2021091423380000.zip

techyminati, 2021-11-18:
> Do I have to combine all super IMGs in the case of MTK firmware such as this? https://fileload.coloros.com/503830RMX2020export_11_C.07_2021091423380000.zip

Yes

huckleberrypie, 2021-11-18:
> Yes

Aight thanks!

huckleberrypie, 2021-11-18:
Also, I've noticed that the combined super.img weighed in at only 6-something gigabytes instead of the 9GB I expected. Doubt that it's compressed either as it isn't packed to hell when I viewed it in a hex editor.