bkerler/oppo_decrypt #43: Encrypt ops doesn't run

vitogga, 2021-06-23:
Every time I try to run encrypt I got usage info. In windows and linux I got the same:
`PS C:\python> python opscrypto.py encrypt C:\python\extract
Usage:
    opscrypto.py --help
    opscrypto.py encryptfile <filename>
    opscrypto.py decryptfile <filename>
    opscrypto.py decrypt <filename>
    opscrypto.py encrypt <directory> [--projid=value] [--firmwarename=name] [--savename=out.ops] [--mbox=version]
Options:
    --projid=value          Set projid Example:18801
    --mbox=version          Set encryption key [default: 5]
    --firmwarename=name     Set firmware version Example:fajita_41_J.42_191214
    --savename=name         Set ops filename [default: out.ops]
PS C:\python> python opscrypto.py encrypt C:\python\extract --projid=19867 --firmwarename=instantnoodlet --savename=out.ops --mbox=6
Usage:
    opscrypto.py --help
    opscrypto.py encryptfile <filename>
    opscrypto.py decryptfile <filename>
    opscrypto.py decrypt <filename>
    opscrypto.py encrypt <directory> [--projid=value] [--firmwarename=name] [--savename=out.ops] [--mbox=version]
Options:
    --projid=value          Set projid Example:18801
    --mbox=version          Set encryption key [default: 5]
    --firmwarename=name     Set firmware version Example:fajita_41_J.42_191214
    --savename=name         Set ops filename [default: out.ops]`
I tried v 1.4 and v 1.3 but not difference.

vitogga, 2021-07-08:
It is not related to this tool.

closed 2021-07-08 by vitogga

SagarMakhar, 2023-05-13:
@vitogga were you able to boot from reencrypted ops ?