bkerler/oppo_decrypt #51: merge the super.img

qianmo-xw, 2022-01-15:
Some OPPO MTK devices ofp file,after unpack ofp file.Will show super0.xxx.img super1.xxx.img，and have a super_map.scv file,this file include super file address. so can we support auto merge all the super file to super.img？
device is oppo a93，if you don't have ofp file. I can provide one for you.

MlgmXyysd, 2022-01-15:
Automatic merging destroys the file structure. You can merge them manually.

qianmo-xw, 2022-01-16:
> Automatic merging destroys the file structure. You can merge them manually.

Which tool can merge them manually,In fact, I just don't know what tools can

MlgmXyysd, 2022-01-16:
> > Automatic merging destroys the file structure. You can merge them manually.
> 
> Which tool can merge them manually,In fact, I just don't know what tools can

https://github.com/bkerler/oppo_decrypt/issues/41#issuecomment-873375685

qianmo-xw, 2022-01-16:
> > > 自动合并会破坏文件结构。您可以手动合并它们。
> > 
> > 
> > 哪个工具可以手动合并，其实我就是不知道什么工具可以
> 
> [第41章（评论）](https://github.com/bkerler/oppo_decrypt/issues/41#issuecomment-873375685)

Ok，Thank you.

closed 2022-01-16 by qianmo-xw