bkerler/oppo_decrypt #56: How was the program developed?

italorecife, 2022-01-20:
@bkerler 
I analyzed the code these days and I understand how it works, I just don't understand how you found out that the packages, ofp for example, were unzipped that way, I also analyzed the package after it was unzipped and there are still some non-00 bytes left between some files , what are these bytes? what are they for?

italorecife, 2022-01-20:
@MlgmXyysd could you answer that too?

MlgmXyysd, 2022-01-20:
Sorry, I don't know. I recently got a batch of accounts, so I haven't studied this for the time being.

bkerler, 2022-01-20:
It's done by reverse engineering and guessing. The original files are protected against reverse engineering. No issue, thus closing this now.

closed 2022-01-20 by bkerler