bkerler/oppo_decrypt #60: help support Repack OFP file

qiurigao, 2022-03-09:
Is it possible to re-encrypt ofp files like ops?

italorecife, 2022-03-09:
I have a personal project that does this but it didn't work in mstool

qiurigao, 2022-03-09:
> I have a personal project that does this but it didn't work in mstool

Where is the problem? verify?
At present, I have received a lot of feedback that the cdt does not exist when I flash the machine. I noticed that there is a switch, and I want to turn it off. need to be repackaged
[image]

MlgmXyysd, 2022-03-11:
I have the same problem

CC @bkerler

qiurigao, 2022-03-12:
> 是否可以像 ops 一样重新加密 ofp 文件？

> Is it possible to re-encrypt ofp files like ops?

@bkerler

italorecife, 2022-03-12:
> > I have a personal project that does this but it didn't work in mstool
> 
> Where is the problem? verify? At present, I have received a lot of feedback that the cdt does not exist when I flash the machine. I noticed that there is a switch, and I want to turn it off. need to be repackaged [image]

when I zip with my program I try to unzip with the tool and it works but if I try to use it in an msmtool it shows an error saying it cannot read the configuration file

qiurigao, 2022-03-12:
> when I zip with my program I try to unzip with the tool and it works but if I try to use it in an msmtool it shows an error saying it cannot read the configuration file

Try to repackage without modification and compare whether the md5 value is the same

italorecife, 2022-03-12:
> Try to repackage without modification and compare whether the md5 value is the same

it won't be because after analyzing the package in a hex editor, there are snippets of data in the package that is not extracted by the program, I believe it could be meta data essential for extracting the flash from the file or something.

Dreamail, 2022-07-28:
will it be supported?