bkerler/oppo_decrypt #66: Unable to extract OnePlus 10 Pro ColorOS OFP file

etherealremnant, 2022-06-19:
When running the process, I get this error.

Traceback (most recent call last):
  File "/mnt/NE2210domestic_11_A.12_2022030701060000/ofp_qc_decrypt.py", line 358, in <module>
    main()
  File "/mnt/NE2210domestic_11_A.12_2022030701060000/ofp_qc_decrypt.py", line 306, in main
    pagesize,key,iv,data=generatekey2(filename)
  File "/mnt/NE2210domestic_11_A.12_2022030701060000/ofp_qc_decrypt.py", line 120, in generatekey2
    pagesize,data=extract_xml(filename,key,iv)
  File "/mnt/NE2210domestic_11_A.12_2022030701060000/ofp_qc_decrypt.py", line 147, in extract_xml
    dec=aes_cfb(data,key,iv)
  File "/mnt/NE2210domestic_11_A.12_2022030701060000/ofp_qc_decrypt.py", line 160, in aes_cfb
    ctx = AES.new(key, AES.MODE_CFB, iv=iv, segment_size=128)
  File "/home/rob/.local/lib/python3.10/site-packages/Crypto/Cipher/AES.py", line 95, in new
    return AESCipher(key, *args, **kwargs)
  File "/home/rob/.local/lib/python3.10/site-packages/Crypto/Cipher/AES.py", line 59, in __init__
    blockalgo.BlockAlgo.__init__(self, _AES, key, *args, **kwargs)
  File "/home/rob/.local/lib/python3.10/site-packages/Crypto/Cipher/blockalgo.py", line 141, in __init__
    self._cipher = factory.new(key, *args, **kwargs)
SystemError: PY_SSIZE_T_CLEAN macro must be defined for '#' formats

File: https://yun.daxiaamu.com/OnePlus_Roms/%E4%B8%80%E5%8A%A010%20Pro/9008%E7%BA%BF%E5%88%B7%E6%95%91%E7%A0%96%E5%8C%85ColorOS%2012.1%20A.12/NE2210domestic_11_A.12_2022030701060000.zip?preview

bkerler, 2022-06-26:
Looks like you need to install the pycryptodome requirements correctly

etherealremnant, 2022-06-26:
I had already done that. I installed it again, getting the exact same error.

It's all good though, no need to keep troubleshooting if you managed to get it to work, must be something with Ubuntu and I don't have the interest in spending the time to fix it since we now know that Qualcomm changed Sahara anyway so my plan wouldn't work. Sorry to waste your time.

closed 2022-06-26 by etherealremnant