bkerler/oppo_decrypt #73: problem with python 3

blueaux75, 2022-08-07:
sorry i'm trying to decrypt a realme firmware i followed all the installation instructions python 3 and pip 3 but something doesn't work i did various tests and in all of them it gives me an error if someone can help me please!!!

PS C:\Users\ameri\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Python 3.10> python ofp_qc_extract.py realme.ofp C:\realme
C:\Users\ameri\AppData\Local\Programs\Python\Python310\python.exe: can't open file 'C:\\Users\\ameri\\AppData\\Roaming\\Microsoft\\Windows\\Start Menu\\Programs\\Python 3.10\\ofp_qc_extract.py': [Errno 2] No such file or directory
PS C:\Users\ameri\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Python 3.10> python3 ofp_qc_decrypt.py realme.ofp C:\realme
Traceback (most recent call last):
  File "C:\Users\ameri\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Python 3.10\ofp_qc_decrypt.py", line 9, in <module>
    from Crypto.Cipher import AES
ModuleNotFoundError: No module named 'Crypto'
[image]

skiddiegit, 2022-08-08:
pip install and then the module which is missing

blueaux75, 2022-08-11:
> pip install and then the module which is missing

I followed the guide in github then: pip3 install -r requirements.txt from the beginning but it doesn't work .... can you help me?

blueaux75, 2022-08-11:
I'm using Windows 11 home

Lanchon, 2022-10-16:
> Don't use Python 3 ...

lol python 2 is unsupported and should not be used. python 3 has existed since around 2005.

https://www.python.org/doc/sunset-python-2/

closed 2023-10-09 by bkerler