bkerler/oppo_decrypt #80: Unknown key. Please ask the author for support :)

Sky84, 2023-03-21:
When i  `python3 ofp_mtk_decrypt.py RMX2086export_11_C.13_2022051901560000.ofp out`

Return : 
```

b'ab3f76d7989207f2' b'2bf515b3a9737835'
Unknown key. Please ask the author for support :)
```
Please help, its for a realme x3 SuperZoom

Arjunr3108, 2024-03-24:
Same problem for realme 3i - 18601_RMX1821EX_11_C.14.ofp, this file was downloaded from realme community how to flash realme 3i tutorial but i once accidentally choose QUALCOMM py file it started decrypting five file and stopped at system.img, when choosing MediaTek py file it immediately throws this error : "b'ab3f76d7989207f2' b'2bf515b3a9737835'
Unknown key. Please ask the author for support :)". Please help '😢

Prinzenglish, 2024-05-17:
same problem Unknown key. Please ask the author for support :) Oppo R11 CPH1707EX_11_A.27_190911.ofp 
I need help please

mu21043, 2024-06-24:
> same problem Unknown key. Please ask the author for support :) Oppo R11 CPH1707EX_11_A.27_190911.ofp I need help please

same error